fotherja/Grid_Tie_Inverter_V5 #1: External pulldown for PWM enable

hydroid7, 2023-03-18:
Hi there,

the state of the IOs is only guaranteed to be stable only after reset. Even then, the pull-up/down resistors are only 40k (see [datasheet](https://static6.arrow.com/aropdfconversion/816a3c5cd038f14b9f633d563429525ba36ff4dc/7dm00172872.pdf), p. 127).

I think adding a 10k pulldown to the `PWM_Disable` would help to keep this pin in a stable (off) state. 
In case it does not required it can be left unpopulated on the PCB.

Cheers!

fotherja, 2023-03-20:
Totally agree with you. I'll add this. Thanks. The PCBs have been shipped and should be with me in about 1 week.

hydroid7, 2023-03-20:
Sure! Do you plan to keep this repo alive or create a new one?

fotherja, 2023-03-27:
Hopefully keep this one alive from now on