fotherja/Grid_Tie_Inverter_V5 #2: Prevent relay arcing with snubber circuit

hydroid7, 2023-03-20:
In case of a forceful diconnect from the grid, currents can flow through the relais, which can lead to building up an arc.

The EMI caused by the arc can cause the microcontroller get into an undefined state e. g. this [StackExchange question].
Therefore it would be nice to include a protection for prevent arcing.
This could be done with RC snubber circuits or with an additional solid state relay, as mentioned in the question.

[StackExchange question]: https://electronics.stackexchange.com/questions/308379/switching-hv-dc-relay-on-crashes-microcontroller

Flole998, 2023-03-20:
RC snubbers are not a good idea as there's still current flowing. With such small currents this shouldn't be an issue at all, I'm switching much larger loads and motors on other PCBs without ever seeing a MCU crash due to that.

hydroid7, 2023-03-21:
Glad to hear that your design works well.

Indeed, the SMA SB2500 uses also 3 relays for this reason.

Flole998, 2023-03-28:
But that's not necessary at all, you simply don't have to switch those under load. As the inverter has FETs aswell for converting DC to AC you simply switch the (unnecessary) relay only after making sure the FETs are off.

Keep in mind that with additional components that permanently consume electricity while the inverter is operating the efficiency goes down.

fotherja, 2023-03-28:
I agree with you Flole998. I did question the necessity of the relays given how much space they take up. I saw them on some GTI reference designs and figured they improved the safety of the system. They will also allow the current sensors to be zeroed more accurately.

fotherja, 2023-03-30:
hydroid7 - I've only just understood your point!

I have realised that if the relays were to open with 6Amps flowing through them then the grid side inductor (1mH) is going to force some horrid voltages. There is also an output fuse and if that were to blow then we'd also get horrid voltages. 

The relays shouldn't switch under load in normal operation but they could... I will add a 400V TVS to the output but that won't fix the potential for arcing when the relays open. I can see an RC snubber fixing this issue well. Don't you agree Flole998?

Flole998, 2023-03-30:
I don't think that actually matters, but even if you add it in case someone doesn't want it they can simply leave it unpopulated.

fotherja, 2023-03-30:
What do you think wouldn't matter? The potential for arcing?

If a relay opened with current flowing would it not create a large (damaging 1000s volts) common mode voltage between the inverter output and the bus voltage?

Flole998, 2023-03-30:
Well if the purpose of the relay is to do galvanic isolation then adding the RC circuit would be against the purpose.

The only reason I can think of why the relay would turn off when current is flowing through it is a DC disconnect under load, which shouldn't normally happen. Even if it does, the FETs should be turning off faster than the relay, so it would not switch under load anyways.

hydroid7, 2023-03-30:
Yes, a TVS would also solve the problem. I think all our statements are partially correct. 

The SMA-SB2500 uses 3 relays, one with a resistance and two for disconnect. Their design is also a little bit different, so it's OK not to include it into this design. 

For example, to do an over current disconnect, you open the relay with the resistance in parallel.
Then to disconnect fully, you open the isolation relais. 

However, since PCB is cheap and as @Flole998 also mentioned, it can be included and marked as DNP.

[image]

Flole998, 2023-03-30:
You could also add a Varistor, that's cheaper (but if it's frequently used it will break at some point).