fotherja/Grid_Tie_Inverter_V5 #3: Schematic update

folny, 2024-01-15:
Hi James

I'm going to try to build your GTI 5 project, but I'm going to modify the PCB to only 2 layers and I want to add ModBus communication to support the smart meter DTSU666 to prevent the flow to the distribution network. Do you have any ideas what I could change and which unnecessary components I can remove from the scheme ?.

fotherja, 2024-01-15:
Hi, Sounds good, I'd be very interested to hear your progress as you go along, keep in touch.  How are you going to be generating your DC bus voltage?

As to possible changes, it depends on what you're aiming for but simplifications that could be considered:
1) Don't bother measuring capacitor current so get rid of the associated 5v isolated supply and the AMC1306
2) Do you want the relays?

Don't forget the TVS.

folny, 2024-01-16:
Instead of measuring the capacitor current, I would like to measure the DC input current, which would be more useful. Furthermore, I am thinking of removing the damping coil L4, I would definitely leave the relay there for safety reasons. I am also thinking about DC Boost at the input, as some professional inverters have.

block diagram Sofar solar https://www.sofarsolar.com/upload/file/20230920/1695196218979040710.pdf

fotherja, 2024-01-16:
The reason for measuring the capacitor current is because it allows you to do active filtering of the LCL resonance in software. That increases efficiency because it allows you to reduce/eradicate the damping resistance in the LCL filter. It's a technique I haven't done yet but I built the inverter in anticipation of trying it out. Don't think any of that can be done by measuring the DC input current.

folny, 2024-01-16:
Yes, I understand that it is not possible by measuring the input current, I had in mind that if the capacitor current measurement still does not work, then I will use the components to measure the input current and calculate the input and output power to calculate the efficiency of the inverter.

folny, 2024-01-16:
Check out this budding project, it looks like it's going to be a great project too.

https://github.com/OpenCleanEnergy/OpenMI

fotherja, 2024-01-16:
That OpenMI project is fantastic! Thank you very much for bringing it to my attention. I shall follow it closely. Keep me updated with your progress too

folny, 2024-01-17:
I am also very curious about the first working prototype of OpenMI. The PCBs for your project will arrive next week so I will test it and let you know how my modified PCB work. My goal is to make an inverter with a power of at least 2000W.