fotherja/Grid_Tie_Inverter_V5 #4: Experimental design(s)

gigaj0ule, 2024-01-27:
Not an issue, but more of a... build log?

Hi there! I'm a power electronics engineer with too much time :-)

This project is awesome and inspired me to build a PCB so I can play around with it too!

[image]

It omits the unnecessary parts (capacitor current measurement, relays). I also simplified the auxiliary power, and re-worked the layout to be more noise immune. 

I haven't installed TVS diodes yet, or ceramic capacitors on VBUS ...I need to dig through the parts catalog and find some appropriate ones. 

Hopefully I can get this working! I'd like to try simpler sensors if I do, such as optocouplers and hall effect sensors. Then, maybe we can use the internal ADC again instead of the expensive external ones.

gigaj0ule, 2024-01-28:
this isn't the right place for a build log i should use instructables

closed 2024-01-28 by gigaj0ule

zwn, 2024-01-28:
Can you add a link to the instructable?

fotherja, 2024-01-28:
https://jamesfotherby.com/index-page/600w-grid-connected-inverter-v5/