fotherja/Grid_Tie_Inverter_V5 #6: I made a new board, but cannot compile the project

gigaj0ule, 2024-02-24:
Hello hackers!

I just built an improved board for this inverter, with simpler layout, and fewer components, and just finished assembling it today!

[image]

However, when I try to compile the stm32 project, I cannot do so in the stm32cubeIDE. 

Since there is no .project file in the repository, I attempt to create my own from the .IOC file, but am encountering errors of all kinds.

Do we still have the .project file? Or, a makefile?

folny, 2024-02-24:
Hi

I also modify this project according to myself and I do the programming in Windows in ARM Keil uVision 5.38 and I have no problem with the compilation. If you decide to use this option, write here and I will give a printscreen of how and what needs to be set in KEIL ARM.

folny, 2024-02-24:
Here is the code set up and compiled in KEIL ARM, hex files can be found in STM32CubeIDE_Code\MDK-ARM\GTI_v5

[STM32CubeIDE_Code.zip](https://github.com/fotherja/Grid_Tie_Inverter_V5/files/14392029/STM32CubeIDE_Code.zip)

folny, 2024-02-24:
I use this program for debugging which is better than debugging in Keil

https://github.com/klonyyy/STMViewer

gigaj0ule, 2024-02-24:
thank you :-)

by any chance, are you able to see the terminal output when compiling?

compiler flags should be included in the debug spew, and although the MDK flags aren’t all going to be the same as GCC ones, I may be able to cross reference them and add to the makefile.

i think the debug output of STM32CUBEIDE should be identical to arm-none-eabi, since it uses GCC internally

folny, 2024-02-24:
I also tried to build the project in STM32CubeIDE and the compilation went without any problems.
[image]